embassy-rs/embassy #4165: SAI read future never completes in `examples/stm32h7/.../sai.rs`

vinsynth, 2025-05-04:
I'm using a Daisy Seed, rev. 7 as required by the example. I replaced the necessary chip definitions in `.cargo/config.toml` and `Cargo.toml`, in addition to adding the `memory.x` file equivalent linked [here](https://github.com/electro-smith/libDaisy/blob/master/core/STM32H750IB_flash.lds) from the manufacturer. As far as I can tell, with this setup, neither `sai_receiver.read(...)` nor `sai_transmitter.write(...)` in `sai.rs` ever complete. I don't think it's a hardware issue; running a passthrough example from the manufacturer on the chip works as expected. Perhaps some defaults have changed for clock configuration since the example's writing, and now the SAI peripheral isn't being initialized as it should be? Or something is wrong with my `memory.x`?

finnh-dev, 2025-05-05:
I am currently investigating a similar problem for the [daisy-embassy](https://github.com/Dicklessgreat/daisy-embassy) BSP. 
The interesting bit is, that our code still works for the Daisy Seed 1.1 version (WM8731 Codec) but encounters the same issue with the 1.2 version (PCM3060 Codec).
From what i've gathered when looking at the differences in SAI configuration between the codecs the main difference is the Master/Slave mode of the transmitter and reciever:
| Codec | SAI Transmitter Mode | SAI Receiver Mode |
|:---|:---|:---|
|WM8731| Slave | Master |
|PCM3060| Master | Slave |

Switching the Master/Slave Configuration for the PCM3060 implementation seems to fix the issue where the future never completes but doesnt produce the expected audio output (In my case just some noticable noise instead of the expected audio passthrough).

We recently switched from using `embassy-stm32` at rev `ca3d091faf3ef76ea45f88eea5c2a6700378daff` to version `0.2.0` (rev ` 6034b17728b3528e42c8499a2893dc35d51d559`) and this seems to cause the issue as using older versions of `daisy-embassy` still works as expected.

Dicklessgreat, 2025-05-05:
Just to add a bit of context,, the Seed 1.1 uses I2C to initialize the WM8731, while the Seed 1.2 does not—it's configured in ”hardware mode” (using its config pins), so the settings are fixed and the i2c interface is not connected like the 1.1. I'm not yet sure how this difference affects the current SAI implementation, but I can at least confirm that the combination of WM8731 and I2C works correctly.

vinsynth, 2025-05-05:
Found the problem--luckily, it's pretty simple. When @Dicklessgreat wrote the original example code, the transmitter was started right after the receiver via an explicit call to `transmitter.start()`; this was later changed as this start call was moved into `transmitter.write()` I think to avoid starting without initializing the DMA buffer; however, because the transmitter is the master in this example, and `transmitter.write()` first occurs after `receiver.read()`, a read is attempted on a slave channel before the master is started, meaning the slave has no clock to reference and hangs forever. Simply swapping those statements fixes the example; I can make a quick pull request for the fix.

Dicklessgreat, 2025-05-06:
Congrats!
I recently updated `embassy-stm32` and noticed that calling `.start()` on the SAI transmitter now causes a panic, so I removed the `tx.start()` call.

In the older version, `.start()` was responsible for initializing the DMA and clock. In the new version, the SAI transmitter's DMA isn't initialized until `write()` is called. This means that on rev 1.2, where the transmitter is the master, if `rx.read().await` happens first in the audio loop, it completely stalls because the clock hasn't started yet.

On the other hand, my rev 1.1 board works fine — probably because the receiver is the master (though I’ve forgotten exactly why it’s set up that way). That causes `rx.start()` to be called before the audio loop begins, initializing the DMA and properly starting the clock, so things don’t get stuck.

This explains a lot — thanks so much for your insight!

closed 2025-05-09 by vinsynth