embassy-rs/embassy #4166: embassy-stm32: Missing TIM4 peripheral in stm32f103c8 Peripherals struct

jmjoy, 2025-05-04:
### Description
The STM32F103C8 microcontroller has a TIM4 peripheral according to the ST datasheet, but it's currently missing in the `Peripherals` struct implementation in embassy-stm32.

### Current situation
Looking at the current documentation at https://docs.embassy.dev/embassy-stm32/git/stm32f103c8/struct.Peripherals.html, the `Peripherals` struct for stm32f103c8 is missing the TIM4 peripheral, which should be available on this chip.

Dirbaio, 2025-05-04:
it's most likely likely picked up by the time driver. Try using another timer by enabling the right Cargo feature, such as time-driver-tim2.

jmjoy, 2025-05-04:
> it's most likely likely picked up by the time driver. Try using another timer by enabling the right Cargo feature, such as time-driver-tim2.

It works, Thanks!

closed 2025-05-04 by jmjoy