embassy-rs/embassy #4179: RP235x Blinky Wifi unexpectedly crashed while running

nisembedded, 2025-05-09:
After I downloaded `examples/rp235x/src/bin/blinky_wifi.rs` it unexpectedly crashed after few tens of seconds without any other information.
Also this code crashed after 3-5 seconds  if I try to build  it by mainline compiler. And if I got the public released version of embassy this code doesn't work anytime.

nisembedded, 2025-05-10:
While I use debugger, device runs more time than usually but also failed on:
```
110.023218 DEBUG iovar set gpioout = [01, 00, 00, 00, 01, 00, 00, 00]
└─ cyw43::control::{impl#3}::set_iovar_v::{async_fn#0} @ /Users/nis/Develop/crates/embassy/cyw43/src/control.rs:571 
110.273691 INFO  led off!
└─ blinky_wifi::____embassy_main_task::{async_fn#0} @ src/bin/blinky_wifi.rs:85  
110.273715 DEBUG iovar set gpioout = [01, 00, 00, 00, 00, 00, 00, 00]
└─ cyw43::control::{impl#3}::set_iovar_v::{async_fn#0} @ /Users/nis/Develop/crates/embassy/cyw43/src/control.rs:571 
110.274066 WARN  len inv mismatch
└─ cyw43::structs::{impl#0}::parse @ /Users/nis/Develop/crates/embassy/cyw43/src/structs.rs:107 
```

nui, 2025-05-13:
RPI Pico 2W board? Check this related [issue](https://github.com/embassy-rs/embassy/issues/3960)

nisembedded, 2025-05-13:
Yep, it more like this behaviour, but for other cases it needs to be more investigated.

i509VCB, 2025-07-22:
Can you give this another try after https://github.com/embassy-rs/embassy/pull/4414 was merged?