embassy-rs/embassy #4189: Can't TCP client connection be created when the interrupt pin of W5500 chip is not soldered?

lifeRobot, 2025-05-13:
Embedded board: rp2040
Examples of usage [ethernet_w5500_tcp_client](https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/ethernet_w5500_tcp_client.rs)

Problem description: 
   1. `embassy_net_wiznet::new` and `embassy_net::new` Created successfully
   2. Both `runners.run().await` are running normally
   3. `socket.connect(host_addr).wait` keeps failing
   4. In terms of hardware, I did not solder the int interrupt pin for w5500

Question:
   1. If there is no interrupt signal, will connect continue to fail?
   2. If it is indeed caused by the disconnection of the interrupt pin, what should I do to ensure that 'connect' can succeed without the interrupt pin

contrast:
   1. I can successfully create a TCP client using CircuitPython [Adafruit_CircuitPython_Wiznet5k](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k), Similarly, there are no solder interrupt pins, please take a look and see if this has any reference value

Dirbaio, 2025-05-13:
Currently `embassy-net-wiznet` requires the interrupt pin, yes. It waits in the main loop for the interrupt to process incoming packets, if it's not connected it'll hang forever.

https://github.com/embassy-rs/embassy/blob/575eab3c60c0d2098f9e9d2a22429aa174b6b968/embassy-net-wiznet/src/lib.rs#L69

It is possible to make it work without the interrupt pin, if you add support to `embassy-net-wiznet`. It'd have to repeatedly poll the chip over SPI to check when packets are received, which would use 100% CPU but should still work. If you do, you can replace the interrupt wait with a `yield_now().await` to ensure other async tasks can still run. A PR for this would be welcome :)

lifeRobot, 2025-11-24:
> Hi! We have an implementation / fork of `Runner` for this :
> 
> /// Polling background runner for the driver.
> ///
> /// You must call `.run()` in a background task for the driver to operate.
> pub struct PollingRunner<'d, C: Chip, SPI: SpiDevice, RST: OutputPin> {
>     mac: WiznetDevice<C, SPI>,
>     ch: ch::Runner<'d, MTU>,
>     _reset: RST,
> }
> 
> impl<'d, C: Chip, SPI: SpiDevice, RST: OutputPin> PollingRunner<'d, C, SPI, RST> {
>     /// Run the driver.
>     pub async fn run(mut self) -> ! {
>         let (state_chan, mut rx_chan, mut tx_chan) = self.ch.split();
>         let mut tick = Ticker::every(Duration::from_millis(250));
>         loop {
>             match select(tick.next(), tx_chan.tx_buf()).await {
>                 Either::First(()) => {
>                     if self.mac.is_link_up().await {
>                         state_chan.set_link_state(LinkState::Up);
>                     } else {
>                         state_chan.set_link_state(LinkState::Down);
>                     }
> 
>                     if let Some(p) = rx_chan.try_rx_buf() {
>                         if let Ok(n) = self.mac.read_frame(p).await {
>                             rx_chan.rx_done(n);
>                         }
>                     }
>                 }
>                 Either::Second(p) => {
>                     self.mac.write_frame(p).await.ok();
>                     tx_chan.tx_done();
>                 }
>             }
>         }
>     }
> }
> 
> 
> /// Create a Wiznet ethernet chip driver for [`embassy-net`](https://crates.io/crates/embassy-net).
> ///
> /// This returns two structs:
> /// - a `Device` that you must pass to the `embassy-net` stack.
> /// - a `PollingRunner`. You must call `.run()` on it in a background task.
> pub async fn new_polling<
>     'a,
>     const N_RX: usize,
>     const N_TX: usize,
>     C: Chip,
>     SPI: SpiDevice,
>     RST: OutputPin,
> >(
>     mac_addr: [u8; 6],
>     state: &'a mut State<N_RX, N_TX>,
>     spi_dev: SPI,
>     mut reset: RST,
> ) -> Result<(Device<'a>, PollingRunner<'a, C, SPI, RST>), InitError<SPI::Error>> {
>     // Reset the chip.
>     reset.set_low().ok();
>     // Ensure the reset is registered.
>     Timer::after_millis(1).await;
>     reset.set_high().ok();
> 
>     // Wait for PLL lock. Some chips are slower than others.
>     // Slowest is w5100s which is 100ms, so let's just wait that.
>     Timer::after_millis(100).await;
> 
>     let mac = WiznetDevice::new(spi_dev, mac_addr).await?;
> 
>     let (runner, device) = ch::new(
>         &mut state.ch_state,
>         ch::driver::HardwareAddress::Ethernet(mac_addr),
>     );
> 
>     Ok((
>         device,
>         PollingRunner {
>             ch: runner,
>             mac,
>             _reset: reset,
>         },
>     ))
> }
> We're hesitant to upstream this because we're not sure about the approach, but it works on our board.
> 
> Thoughts ?

Thank you for your suggestion, this is useful, but I used another method without destroying the embassy-net-wiznet library. Based on the method of implementing Wait trait, refer to [wait.rs](https://github.com/lifeRobot/embassy-plus/blob/main/embassy-socket/src/wait.rs)