embassy-rs/embassy #4190: [rp2040] DMA: Enhance PIO RX DMA support for continuous hardware-chained ping-pong buffering via CHAIN_TO

jancespivo, 2025-05-13:
Current high-level transfer API is not designed to support ping-pong buffering using CHAIN_TO.

It is crucial when dealing for example with camera data, where it is needed to fill a DMA buffer with a few lines and processing it very quickly while the second buffer is filled meanwhile.

I've tried double buffer approach without chaining:

```rust
let transfer_a = rx_fifo.dma_pull(dma_channel.reborrow(), buffer_a, false);
transfer_a.await;
  
loop {        
      {
          let transfer_b = rx_fifo.dma_pull(dma_channel.reborrow(), buffer_b, false);
          process_data(buffer_a);
          transfer_b.await;
      }
      {
          let transfer_a = rx_fifo.dma_pull(dma_channel.reborrow(), buffer_a, false);
          process_data(buffer_b);
          transfer_a.await;
      }
}
```
However it is not reliable and I observe missing data quite quickly (it probably depends on other running tasks).

holly-hacker, 2025-10-21:
I need a similar ping-pong buffering strategy where a simple ring buffer wont work. I'll have to look more into how embassy works under the hood (this is my first serious embedded project) but if I manage to cobble something together, I'll open a PR to see it can be upstreamed.

Quick-Flash, 2026-08-27:
A dma ringbuffer would also be very nice to have. Just ran into the fact that I cannot natively set this up in embassy for the rp2350.