embassy-rs/embassy #4200: UdpSocket, W5500: Reading ethernet packets larger than MTU.

kaydensigh, 2025-05-14:
Using the `ethernet_w5500_udp.rs` example for `embassy-rp`, on a [W5500-evb-pico](https://docs.wiznet.io/Product/iEthernet/W5500/w5500-evb-pico), I'm unable to receive packets larger than ~1.5 kB. Packets are sent from a Windows machine. `UdpSocket::read_from` doesn't seem to notice the larger packets, but continues to work for subsequent smaller packets.

I suspect this is something to do with the packet itself reporting a size larger than the MTU, but the W5500's buffer only contains the first fragment. I couldn't figure out if this is a bug, or just required some configuration on the device or stack. Any advice is appreciated, thanks.

Dirbaio, 2025-05-14:
if you're using jumbo frames (ethernet packets actually larger than 1.5k), check if the w5500 supports it, and check you've configured the MTU to be larger. (you might need to modify embassy-net-wiznet for this)

If you're using fragmentation (splitting the IP packet into multiple), check the relevant Cargo features are enabled in smoltcp and that there's a large enough reassembly buffer.

kaydensigh, 2025-05-15:
Thank you so much! Turns out I needed to import `smoltcp` and enable the following features:

```
smoltcp = { version = "0.12.0", default-features = false, features = [
    "proto-ipv4-fragmentation",
    "reassembly-buffer-size-4096",
] }
```

For future readers: on Windows, in device manager, the network adapter properties shows whether jumbo frames is enabled. It's likely disabled by default.

closed 2025-05-15 by kaydensigh