embassy-rs/embassy #4247: embassy-boot-stm32 possibly outdated in crates.io

atan2l, 2025-05-24:
It appears that the `embassy-boot-stm32` crate on crates.io needs an update since it was last updated over a year ago and the current version in the repository, while still `0.2.0` now depends on `embassy-stm32 0.2.0`.

When including it as part of my project's dependencies it causes the build to fail as it imports `embassy-stm 0.1.0`, for which I haven't defined the corresponding `stm32xx` feature.

```
error: failed to run custom build command for `stm32-metapac v15.0.0`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.
Caused by:
  process didn't exit successfully: `/home/eltu/dev/stm32/f767zi-bootloader/target/debug/build/stm32-metapac-f657135d6c3b8a7e/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at /home/eltu/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/stm32-metapac-15.0.0/build.rs:12:10:
  No stm32xx Cargo feature enabled
  stack backtrace:
     0: __rustc::rust_begin_unwind
     1: core::panicking::panic_fmt
     2: core::option::expect_failed
     3: core::option::Option<T>::expect
     4: build_script_build::main
     5: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

atan2l, 2025-05-26:
> You need to use git version of embassy, in order to that, add following section to your Cargo.toml after dependencies section:
> 
> ```
> [patch.crates-io]
> embassy-stm32 = { git = "https://github.com/embassy-rs/embassy", branch = "main" }
> embassy-boot-stm32 = { git = "https://github.com/embassy-rs/embassy", branch = "main" }
> 
> etc...
> ```
> 
> You will have to add it for each embassy dependency you are using in your project.

Hey, thank you for your reply.

And yeah, I ended up referencing the github version directly. I just thought I'd point it out since I didn't see any other issues regarding this and perhaps it was an oversight. If you guys are aware and have decided that referencing the github version is the way to go I'm more than happy to close the issue :)

iamadahn, 2025-05-26:
You are welcome, but I have nothing to do with the devs, I was just browsing through issues and noticed this, which I have encountered too, so I just wanted to help by sharing how I was able to fix that issue, may be there is another way, but I doubt that we will get the answer here, since there are a lot of another issues with no comments. I believe that the place to ask questions like this is their matrix chat

lulf, 2025-05-26:
This was an oversight, thanks for reporting the issue @atan2l . I will release the embassy-boot-stm32 0.3.0 and close this issue when done.