embassy-rs/embassy #4250: nRF TWIM: Potential bug.

Grazfather, 2025-05-24:
I have been using your TWIM implementation as a reference for a HAL for [microzig](https://github.com/ZigEmbeddedGroup/microzig).

I was wracking my brain trying to figure out why I couldn't get a certain sensor to work properly when using `blocking_write` to configure, then `blocking_write_read`. I noticed that the master was not properly NACKing after reading its expected bytes.

Turns out that the issue was that the SHORTS setup in `blocking_write` were never cleared, so they interacted strangely with the ones setup for write-then-read.

I fixed the issue by clearing all shorts before starting a new transaction.

Note that this is a bug in my zig code! I have not even tried to compile anything embassy, but I suspect that you guys have the same issue.

## Repro:
(on nRF52840, to a TMP117)
* write 2 bytes
* write-then-read 1 then 2

## Expected behaviour:
* After reading 2 bytes, master NACKS and stops the transaction

## Actual behaviour
* Master ACKs and keeps reading. In my case the device doesn't send anything, but the master keeps clocking indefinitely.

I don't know rust well, so you guys might be covering this, but I did not see anywhere where you clear out the SHORTS register between operations.

I hope this helps!

lulf, 2025-05-26:
Is this the behavior you see with the Rust example or your zig code? From what I understand, the shorts().write() will clear & set the register, not leaving existing bits in place. And the code seems to always write shorts before starting the transaction.

Grazfather, 2025-05-26:
If that is the behaviour, then I am wrong! I figured it was doing a read-modify-write, not just wiping out the other fields. If that's the case, then I am mistaken! Sorry about that.