embassy-rs/embassy #4332: provide way to get UID as `&str`

rursprung, 2025-06-23:
for both the STM32 and RP2040 the UID is returned as an `[u8]`. the [`embassy_stm32::uid` module](https://github.com/embassy-rs/embassy/blob/454a5e9044dd0b405bc25ff2f05a576939204689/embassy-stm32/src/uid.rs) provides a helper to convert from `[u8]` to `&'static str`.

for RP2040 no such helper exists and similar code patterns are being repeated in various projects, for example:
* https://github.com/OneVariable/poststation-util/blob/de27d7f8e9d097a67ad06d971ff3e1c5c141f3e6/templates/rp2040/src/main.rs#L46-L65
* https://github.com/bugadani/fruitfly/blob/28988827c7e5469714375c210d2c23cb0b730d50/src/main.rs#L56-L73

it'd be great if this could somehow be resolved centrally, so that for RP2040 (and RP2350, i guess?) a similarly simple API exists as for STM32, which directly returns an `&'static str`

leftger, 2026-10-09:
This already exists — `embassy_stm32::uid::uid_hex()` returns `&'static str`.

In `embassy-stm32/src/uid.rs`, `uid_hex()` is documented as "this device's unique 96-bit ID, encoded into a string of 24 hexadecimal ASCII digits", alongside `uid() -> &'static [u8; 12]` and `uid_hex_bytes() -> &'static [u8; 24]`.

(Note this predates the issue — it's been there since Dec 2023 — so it may just have been easy to miss. Reopen if you wanted a different representation.)

rursprung, 2026-10-10:
@leftger: this issue is about RP2040, not STM32. my initial comment shows that STM32 has the API but RP2040 does not and asks for it to be added to RP2040 as well

leftger, 2026-10-10:
Ok I've reopened it

Dirbaio, 2026-10-11:
it does exist in RP now though :) 

https://github.com/embassy-rs/embassy/pull/7083

closed 2026-10-11 by Dirbaio

rursprung, 2026-10-13:
ah, nice! thanks for pointing that out @Dirbaio!

would it make sense to add an abstraction for that in `embassy-embedded-hal` (or even `embedded-hal`?) so that in the future there's a generic way to get the UID of a device? or is this just too different across device types to try to abstract it away?