embassy-rs/embassy #4367: `embassy-executor`: High RAM usage

davidedellagiustina, 2025-07-03:
While experimenting on an nRF52840, I quickly got stuck at a stack overflow as soon as I got a couple of non-trivial tasks running on the `embassy-executor`. Upon further inspection, it seems that the stack pointer that I get inside any task is almost at the beginning of the RAM section, which means that it may very quickly overflow.

Here's a trivial example that uses the trick of allocating a zero-sized struct to retrieve and print the addrress of the top of the stack:

```rust
use {defmt_rtt as _, panic_probe as _};

static EXECUTOR: static_cell::StaticCell<embassy_executor::Executor> = static_cell::StaticCell::new();

#[cortex_m_rt::entry]
fn main() -> ! {
    let peripherals = embassy_nrf::init(embassy_nrf::config::Config::default());
    let executor = EXECUTOR.init(embassy_executor::Executor::new());

    struct Dummy {}
    let x = Dummy {};
    defmt::info!(
        "Top of stack outside task: {:#x}",
        (&x) as *const Dummy as usize
    );

    executor.run(|spawner| {
        defmt::unwrap!(spawner.spawn(dummy_task()));
    })
}

#[embassy_executor::task]
async fn dummy_task() {
    struct Dummy {}
    let x = Dummy {};
    defmt::info!(
        "Top of stack inside task: {:#x}",
        (&x) as *const Dummy as usize
    );

    core::future::pending::<()>().await;
}
```

This example produces the following output for me (on a chip where `0x20000000` marks the beginning of the RAM section):
```
0.000213 [INFO ] Top of stack outside task: 0x2003ffef
0.000274 [INFO ] Top of stack inside task: 0x20005c59
```

Is this intended behavior? If so, how to get around this and allow tasks to use more stack space?

davidedellagiustina, 2025-07-03:
It makes sense, I'll give it a try tomorrow as soon as I get back on the code and report my findings here.

However, this leaves my original point unanswered: where could the stack overflows I see come from?
Actually, I'm not even sure they're stack overflows: what I see is random panics which don't happen if I comment out some task (I have two at the moment) or swich compilation profiles, which to me feels like something's being overwritten in memory, potentially as a result of a stack overflow.
Is it just that my futures are too big? How can I check this hypothesis, and how can I reduce their sizes?

I've worked with ESP32 boards in the past running Rust code using ESP-IDF (i.e. bindings to FreeRTOS, which should be a much more RAM-hungry approach) instead of Embassy, and I've never really experienced stack overflows even while doing pretty complicated things using tasks with as little as 10-12KiB of stack, whereas now I'm running into issues while using no RTOS and having ~256KiB of RAM available, so I really can't figure out what's going on.

Dirbaio, 2025-07-03:
you can objdump to see the task futures in statics, and where .bss ends. Space between end of .bss and end of ram is spce left for the (non async) stack. 
If .futures are bigger than you expect, -Zprint-type-sizes prints their size and why. A common source of bloat is passing big things by value as args to async fns.

davidedellagiustina, 2025-07-04:
My findings from today:
- `cortex_m::register::msp::read()` always show me addresses in the `0x2003fXXX` range, which look correct since I shouldn't be using that much stack upon a more detailed code inspection;
- this is my `.bss` section when compiling with `--release`, as reposted by `objdump -h`:

```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
[...]
6 .bss          00001df0  20005a60  20005a60  0000b640  2**2
                  ALLOC
[...]
```
It looks like it is actually not that big, and shouldn't collide with the stack addresses I see. If I understand correctly, it should span the `0x20005a60 - 0x20007850` range.
- this is the exact panic message I see, which I just assumed to be a stack overflow due to the randomness of the phenomenon:

```
Firmware exited unexpectedly: Multiple
Core 0
    Frame 0: _defmt_encoding_ = rzcobs @ 0x00000a60> @ 0x0000000000000a60

    Frame 1: dequeue_all<embassy_executor::raw::{impl#9}::poll::{closure_env#0}> @ 0x0002e85c inline
       /home/ddgiustina/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/run_queue_atomics.rs:78:19
    Frame 2: poll @ 0x000000000002e83a inline
       /home/ddgiustina/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/mod.rs:423:9
    Frame 3: poll @ 0x000000000002e83a
       /home/ddgiustina/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/mod.rs:533:9
```

davidedellagiustina, 2025-07-04:
Additional info that I just discovered: my panics seem to be correlated with how fast I log stuff via `defmt-rtt`. The same happens also with `rtt-target` though. Maybe there's too much stuff queuing up?

davidedellagiustina, 2025-07-15:
@golpan-rs @Dirbaio 

So, here's a small(-ish) program that exhibits the weird crash I described. It's not super compact, but it's the best I could do without completely changing what I'm trying to actually do.
It uses the NRF softdevice S140 to start a Bluetooth scan on one of the two tasks. The softdevice can be flashed with `probe-rs` following the instructions here: https://github.com/embassy-rs/nrf-softdevice.

Note that the crash happens only if both tasks are running, while it does not happen if any of the two is commented out.

This is my `memory.x` file:
```
MEMORY
{
    FLASH (rx) : ORIGIN = 0x0 + 156K, LENGTH = 1024K - 156K
    RAM (rwx) : ORIGIN = 0x20000000 + 0x5a08, LENGTH = 256K - 0x5a08
}

_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

And this is the `main.rs` of the program:
```rust
#![no_std]
#![no_main]

extern crate alloc;

use alloc::boxed::Box;
use alloc_cortex_m::CortexMHeap;
use embassy_executor::Spawner;
use embassy_nrf::{Peri, config, interrupt, pac, peripherals, usb};
use embassy_usb::class::cdc_acm;
use nrf_softdevice::{Softdevice, ble::central, raw};
use {defmt_rtt as _, panic_probe as _};

embassy_nrf::bind_interrupts!(
    pub struct Irqs {
        USBD => usb::InterruptHandler<peripherals::USBD>;
        CLOCK_POWER => usb::vbus_detect::InterruptHandler;
    }
);

#[global_allocator]
static HEAP: CortexMHeap = CortexMHeap::empty();

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    defmt::info!("Starting...");

    {
        use core::mem::MaybeUninit;
        const HEAP_SIZE: usize = 20 * 1024; // Bytes
        static mut MEM: [MaybeUninit<u8>; HEAP_SIZE] = [MaybeUninit::uninit(); HEAP_SIZE];
        unsafe { HEAP.init(&raw mut MEM as usize, HEAP_SIZE) };
    }

    defmt::trace!("Initializing peripherals...");
    let mut config = config::Config::default();
    config.time_interrupt_priority = interrupt::Priority::P2;
    config.gpiote_interrupt_priority = interrupt::Priority::P2;
    let peripherals = embassy_nrf::init(config);

    defmt::unwrap!(spawner.spawn(usb_task_run(peripherals.USBD)));
    defmt::unwrap!(spawner.spawn(bluetooth_task_run(spawner)));

    core::future::pending::<()>().await;
}

#[embassy_executor::task]
pub async fn usb_task_run(peripheral: Peri<'static, peripherals::USBD>) {
    defmt::trace!("Initializing USB task...");

[... 91 lines cut ...]
    loop {
        let config = Box::new(central::ScanConfig {
            extended: true,
            interval: 1600, // 1s = 1600 * 625us
            timeout: 0,
            ..Default::default()
        });

        defmt::info!("Starting device scan...");
        central::scan(softdevice, &config, |args| {
            defmt::info!("Discovered device: {:#04x}", args.peer_addr.addr);
            Option::<()>::None
        })
        .await
        .unwrap();
    }
}

#[embassy_executor::task]
async fn softdevice_rask_run(softdevice: &'static Softdevice) {
    softdevice.run().await;
}
```

Let me know if you need anything else to try to reproduce the issue.

davidedellagiustina, 2025-07-15:
@Dirbaio Thank you for your quick feedback.

So, I am already using the `critical-section-impl` from the  `nrf-softdevice` crate, so that should be ok.
Intreeupt priorities should also be ok: I already followed the instructions you linked as you can see from the code I posted.

However, I have to admit that I'm not entirely sure how the `HardwareVbusDetect` thing works -- I took it directly from the embassy examples (https://github.com/embassy-rs/embassy/blob/main/examples/nrf52840/src/bin/usb_serial.rs). I'll try to investigate in that direction, thank you for the pointer.

Let me know if there's anything else I should double-check.

Dirbaio, 2025-07-15:
> Intreeupt priorities should also be ok

you're not setting the priorities of the USBD and CLOCK_POWER interrupts (though you shouldn't touch CLOCK_POWER directly because of the other issue)

davidedellagiustina, 2025-07-16:
@Dirbaio Do you have any reference or example on how to set the USBD interrupt priorities?