embassy-rs/embassy #4371: [Feature Request] Implement missing I2C transaction methods for STM32 I2C v2 driver

HybridChild, 2025-07-05:
## Summary

The STM32 I2C v2 driver (`embassy-stm32/src/i2c/v2.rs`) currently has `todo!()` placeholders for the `blocking_transaction()` and `transaction()` methods, making it non-compliant with the embedded-hal I2C trait requirements. This prevents users from performing multi-operation I2C transactions that are essential for many I2C devices drivers.

## Current State

```rust
pub fn blocking_transaction(&mut self, addr: u8, operations: &mut [Operation<'_>]) -> Result<(), Error> {
    let _ = addr;
    let _ = operations;
    todo!()
}

pub async fn transaction(&mut self, addr: u8, operations: &mut [Operation<'_>]) -> Result<(), Error> {
    let _ = addr;
    let _ = operations;
    todo!()
}
```

## Problem description
The missing implementation prevents compliance with embedded-hal I2C trait contract.

The [embedded-hal transaction contract](https://docs.rs/embedded-hal/latest/embedded_hal/i2c/trait.I2c.html#tymethod.transaction) requires:

- START condition before first operation
- Data merging for consecutive same-type operations (no intermediate START/STOP)
- RESTART between different operation types
- STOP condition only after the final operation
- Proper NACK handling on last read byte

## Proposed Solution
Implement both methods using STM32 I2C v2's hardware-assisted features.

### Architecture

- **Group consecutive same-type operations** for contract-compliant merging
- **Leverage RELOAD mechanism** for seamless operation continuation
- **Use RESTART capability** for proper transitions between operation types
- **Coordinate AUTOEND behavior** for correct STOP condition placement

## Implementation Status
I have an implementation that has been analyzed for contract compliance and hardware compatibility. Next step is to thoroughly test the implementation. I plan to do this by writing test firmware for two stm32 chips (i2c master and slave) that send data back and forth in various configurations to verify contract compliance.

HybridChild, 2025-07-05:
@Dirbaio - I'd like to implement I2C v2 transaction support as described above. 

A few questions:
- Should I split blocking vs async into separate PRs?
- Any specific testing requirements for STM32 I2C changes?

Happy to refine the approach based on your feedback!

tegimeki, 2025-07-14:
This is something I need as well, for talking to LM92 temperature sensors from an H723.  So if I can help test (or collaboratively implement) let me know.

closed 2025-11-20 by HybridChild