embassy-rs/embassy #4461: embassy-rp stuck in pre_init on wokwi RP2040 emulator

manobendro, 2025-07-25:
I try to run embasy-rp blinky example in wokwi RP2040 emulator, but it stuck in pre_init(), examining with gdb.

How to re-produce

`cd examples/rp`
`cargo build --bin blinky`
`cargo run --bin blinky`

also, I change runner elf2uf2-rs with removing -d flag to generate an uf2 file

wokwi.toml
```wokwi.toml
[wokwi]
version = 1
firmware = 'examples/rp/target/thumbv6m-none-eabi/debug/blinky.uf2'
elf = 'examples/rp/target/thumbv6m-none-eabi/debug/blinky'
gdbServerPort=3333
```
diagram.json
```diagram.json
{
  "version": 1,
  "author": "Manob Biswas",
  "editor": "wokwi",
  "parts": [
    {
      "type": "wokwi-pi-pico",
      "id": "pico",
      "top": 0,
      "left": 0,
      "attrs": { "builder": "pico-sdk" }
    }
  ],
  "connections": [ [ "pico:GP0", "$serialMonitor:RX", "", [] ], [ "pico:GP1", "$serialMonitor:TX", "", [] ] ],
  "dependencies": {}
}
```
GDB screenshot

[image]

Dirbaio, 2025-07-25:
not without patchiing Embassy or implementing the missing functionality in the emulator.

manobendro, 2025-07-25:
At least I got some idea.
Thanks

closed 2025-07-25 by manobendro