embassy-rs/embassy #4474: `embassy-rp`: allow zero-length I2C writes

rursprung, 2025-07-28:
currently, the (blocking) I2C implementation in `embassy_rp` does not allow zero-length writes:
https://github.com/embassy-rs/embassy/blob/0180c8067656f37a601fc6e9672707e46646b86a/embassy-rp/src/i2c.rs#L551-L553

however, zero-length writes can/should be used to scan if a device is present on an I2C address. see e.g. [this arduino example](https://learn.adafruit.com/scanning-i2c-addresses/arduino) which just starts and stops an I2C transaction.
just removing the check is not enough as it then does not start and stop an I2C transaction which is needed.

if there's another way to do this that'd of course also be ok, but as discussed today in the Rust Embedded matrix channel a zero-length write should be the correct thing to do (and is portable because it can be done with the `embedded-hal` traits).

this can be used as an example to test if it works (put it in `examples/rp/src/bin/i2c_blocking_scanner.rs`):
```rust
//! This example shows how to write an I2C scanner.
//! Besides the setup code this is portable and written against `embedded-hal` v1.

#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_rp::i2c::{self, Config};
use embassy_time::{Duration, Timer};
use embedded_hal_1::i2c::I2c;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_rp::init(Default::default());

    let sda = p.PIN_0;
    let scl = p.PIN_1;

    info!("set up i2c");
    let mut i2c = i2c::I2c::new_blocking(p.I2C0, scl, sda, Config::default());

    loop {
        i2c_scan(&mut i2c);
        info!("done scanning! will repeat again in 10s");
        Timer::after(Duration::from_secs(10)).await;
    }
}

fn i2c_scan<I2C: I2c>(i2c: &mut I2C) {
    for addr in 1..=127 {
        trace!("scanning address {=u8:#02x}", addr);
        if i2c.read(addr as u8, &mut []).is_ok() {
            info!("found device on address {=u8:#02x}", addr);
        } else {
            debug!("no device on address {=u8:#02x}", addr);
        }
    }
}
```

when you remove the check mentioned above it will not fail but won't find any devices even if they are connected.

rursprung, 2025-07-29:
thanks a lot for digging that up, @CBJamo!

then i guess it's the same question as for `rp-hal`: should a software impl be provided for 0-length writes (e.g. as a wrapper I2C type which supports this, as suggested in the linked issue?)

and i guess the same documentation as there should be added to the errors? https://github.com/rp-rs/rp-hal/pull/679

CBJamo, 2025-07-29:
Docs are always good, so a pr there wouldn't go amiss.

Bitbanging or PIO-ing the zero-length write would be useful in some cases, but I've never seen an i2c part that behaves badly when sent a 1-byte read to probe. It's slower, but probing is usually only a startup procedure anyway.

closed 2026-10-16 by xoviat