embassy-rs/embassy #4489: Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor. After the executor was updated to 0.8 from 0.7

zayats1, 2025-08-01:
<details>
  <summary>
dependencies
  </summary>

cortex-m = { version = "0.7.7", features = ["inline-asm"] }
cortex-m-rt = "0.7.5"
defmt = "1.0.1"
defmt-rtt = "1.0.0"
panic-probe = { version = "1.0.0", features = ["print-defmt"] }
cortex-m-semihosting = "0.5.0"
critical-section = "1.1"
embassy-rp = { version = "0.6.0", features = [
    "defmt",
    "time-driver",
    "critical-section-impl",
    "rp2040",
    "boot2-generic-03h",
] }
embassy-executor = { version = "0.8.0", features = [
    "arch-cortex-m",
    "executor-thread",
    "executor-interrupt",
    "timer-item-payload-size-8",
] }
embassy-embedded-hal = { version = "0.3.1", features = ["defmt"] }
embassy-sync = { version = "0.7.0", features = ["defmt"] }
embassy-futures = "0.1.1"
embassy-time = "0.4.0"
rtt-target = { version = "0.6.1", features = ["defmt"] }
embedded-hal = "1.0.0"

</details>

<details> 
   <summary>
     Logs
   </summary>
   Finished `dev` profile [optimized + debuginfo] target(s) in 0.08s
     Running `probe-rs run --chip rp2040 target/thumbv6m-none-eabi/debug/hello`
      Erasing ✔ 100% [####################]  20.00 KiB @  51.98 KiB/s (took 0s)
  Programming ✔ 100% [####################]  20.00 KiB @  44.88 KiB/s (took 0s)                                                                                                                                                                                                    Finished in 0.83s
[INFO ] Hello world (hello src/bin/hello.rs:21)
[INFO ] led on! (hello src/bin/hello.rs:23)
[ERROR] panicked at /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/waker.rs:38:9:
Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor. (panic_probe panic-probe-1.0.0/src/lib.rs:104)
Firmware exited unexpectedly: Exception
 WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(11002927): An error occurred in the communication with an access port or debug port.
Core 0
    Frame 0: HardFault_ @ 0x10003934
       /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cortex-m-rt-0.7.5/src/lib.rs:1103:1
    Frame 1: HardFault @ 0x1000045a

    Frame 2: __udf @ 0x1000045a inline
       /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cortex-m-0.7.7/src/../asm/inline.rs:181:5
    Frame 3: udf @ 0x0000000010000430 inline
       /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cortex-m-0.7.7/src/call_asm.rs:11:43
    Frame 4: hard_fault @ 0x0000000010000430 inline
       /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/panic-probe-1.0.0/src/lib.rs:86:5
    Frame 5: panic @ 0x0000000010000430
       /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/panic-probe-1.0.0/src/lib.rs:54:9
    Frame 6: {"package":"defmt","tag":"defmt_prim","data":"{=usize}","disambiguator":"5508678152672316438","crate_name":"defmt"} @ 0x00000002> @ 0x0000000000000002

    Frame 7: _defmt_release @ 0x10002046
       /home/bogdandev/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-rtt-1.0.0/src/lib.rs:67:24
    Frame 8: _defmt_encoding_ = rzcobs @ 0x070e0000> @ 0x00000000070e0000

Core 1
    Frame 0: _defmt_encoding_ = rzcobs @ 0x00000184> @ 0x0000000000000184

    Frame 1: _defmt_encoding_ = rzcobs @ 0x00400000> @ 0x0000000000400000
</details>

zayats1, 2025-08-01:
> Until the new embassy-time-queue-utils can be published, you have to add the following in your Cargo.toml:
> 
> ```
> [patch.crates-io]
> embassy-time-queue-utils = { git = "https://github.com/embassy-rs/embassy", rev="e818e49d7a3c27a237bcd6d84df7971c3a02deba"}
> ```
> 
> [@Dirbaio](https://github.com/Dirbaio) The crate is owned by you only, so please add the crates team to it when you have the chance.

thanks!

leocassarani, 2025-08-03:
I had the same issue with an STM32 target, but I was able to fix it by patching a few more crates to use the git revision suggested by @lulf. This was the minimal set of patches that made the error go away for me:

```
[patch.crates-io]
embassy-executor = { git = "https://github.com/embassy-rs/embassy", rev = "e818e49d7a3c27a237bcd6d84df7971c3a02deba" }
embassy-stm32 = { git = "https://github.com/embassy-rs/embassy", rev = "e818e49d7a3c27a237bcd6d84df7971c3a02deba" }
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev = "e818e49d7a3c27a237bcd6d84df7971c3a02deba" }
embassy-time-queue-utils = { git = "https://github.com/embassy-rs/embassy", rev = "e818e49d7a3c27a237bcd6d84df7971c3a02deba" }
```

zayats1, 2025-08-05:
The issue is present in stm32

closed 2025-08-05 by zayats1

lucasw, 2025-08-08:
@leocassarani I tried e818e49d7a3c27a237bcd6d84df7971c3a02deba but still go the error, but f2be66a5f94a655696407e2e7bc81c322aab3ea1 (which is the latest commit here to main as of now) is working for me (and my `stm32h753zi`)

```
[patch.crates-io]
embassy-executor = { git = "https://github.com/embassy-rs/embassy", rev = "f2be66a5f94a655696407e2e7bc81c322aab3ea1" }
embassy-stm32 = { git = "https://github.com/embassy-rs/embassy", rev = "f2be66a5f94a655696407e2e7bc81c322aab3ea1" }
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev = "f2be66a5f94a655696407e2e7bc81c322aab3ea1" }
```

(https://github.com/lucasw/misc_rust/blob/main/nucleo_embassy/Cargo.toml)