embassy-rs/embassy #4511: embassy-stm32 f3: hard fault while using embassy_time::Timer in spawned task.

burumdev, 2025-08-05:
Tried spawning a task which reads temperature data from LSM303AGR magnetometer-accelerometer every 3 seconds:

```rust
#![no_main]
#![no_std]

use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::{
    bind_interrupts,
    i2c::{self, I2c},
    mode::Async,
    peripherals,
    time::Hertz,
};

use embassy_time::Timer;

use lsm303agr::{Error as Lsm303agrError, Lsm303agr, interface::I2cInterface, mode::MagOneShot};

use defmt_rtt as _;
use panic_probe as _;

bind_interrupts!(struct Irqs {
    I2C1_EV => i2c::EventInterruptHandler<peripherals::I2C1>;
    I2C1_ER => i2c::ErrorInterruptHandler<peripherals::I2C1>;
});

type Magnetometer = Lsm303agr<I2cInterface<I2c<'static, Async>>, MagOneShot>;

async fn read_temperature(lsm303agr: &mut Magnetometer) {
    match lsm303agr.temperature().await {
        Ok(temperature) => {
            info!("Temperature reading successful.");
            info!(
                "Current temperature is: {} °C",
                temperature.degrees_celsius()
            );
        }
        Err(err) => {
            let err_txt = match err {
                Lsm303agrError::Comm(_) => "I2C communication error.",
                Lsm303agrError::InvalidInputData => "Invalid input data.",
            };
            error!("ERROR reading temperature: {}", err_txt);
        }
    }
}

#[embassy_executor::task]
async fn read_temperature_every_3_seconds(mut lsm303agr: Magnetometer) {
    info!("*** READING TEMPERATURE THE FIRST TIME ***");
    read_temperature(&mut lsm303agr).await;

    loop {
        Timer::after_secs(3).await;
        info!("*** READING TEMPERATURE EVERY 3 SECONDS ***");
        read_temperature(&mut lsm303agr).await;
    }
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    info!("Does defmt work? Yes it does!");

    let peris = embassy_stm32::init(Default::default());

    let i2c = I2c::new(
        peris.I2C1,
        // These pins are hardwired to the onboard magnetometer LSM303AGR on the stm32f3 discovery
        peris.PB6,
        peris.PB7,
        Irqs,
        // DMA channels for I2c TX DMA (Sender) and RX DMA (Receiver)
        peris.DMA1_CH6,
        peris.DMA1_CH7,
        // We use normal transmission speed
        Hertz(100_000),
        Default::default(),
    );

    let lsm303agr = Lsm303agr::new_with_i2c(i2c);

    spawner
        .spawn(read_temperature_every_3_seconds(lsm303agr))
        .unwrap();
}
```

I get this hard fault error:

> 0.000 [INFO ] Does defmt work? Yes it does! (stm32f3_discovery_compass_embassy stm32f3-discovery-compass-embassy/src/main.rs:61)
> 0.001 [INFO ] *** READING TEMPERATURE THE FIRST TIME *** (stm32f3_discovery_compass_embassy stm32f3-discovery-compass-embassy/src/main.rs:49)
> 0.003 [ERROR] panicked at /home/baris/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/waker.rs:38:9:
> Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor. (panic_probe panic-probe-1.0.0/src/lib.rs:104)
> Firmware exited unexpectedly: Multiple
> Core 0
>     Frame 0: HardFault_ @ 0x08014cc4
>        /home/baris/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cortex-m-rt-0.7.5/src/lib.rs:1103:1
>     Frame 1: HardFault <Cause: Escalated UsageFault <Cause: Undefined instruction>> @ 0x08012d00
> 
>     Frame 2: __udf @ 0x08012d00 inline
>        ./asm/lib.rs:48:1
>     Frame 3: __udf @ 0x0000000008012d00
>        ./asm/lib.rs:51:17
>     Frame 4: disable @ 0x080061ac
>        /home/baris/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cortex-m-0.7.7/src/interrupt.rs:37:1

Isn't embassy executor created in main?

lulf, 2025-08-06:
See https://github.com/embassy-rs/embassy/issues/4507#issuecomment-3156060712 for a workaround until embassy-stm32 is released.

burumdev, 2025-08-06:
Downgrading `embassy-executor` to `0.7.0` version resolved the issue.

I rolled back the git revision setup mentioned earlier and downgraded embassy-executor to 0.7.0. Relevant portion of Config.toml of the working setup is:

```
embassy-executor = { version = "0.7.0", features = ["arch-cortex-m", "executor-thread", "executor-interrupt", "defmt"] }
embassy-stm32 = { version = "0.2.0", features = ["stm32f303vc", "memory-x", "time-driver-tim2"] }
embassy-sync = "0.7.0"
embassy-time = { version = "0.4.0" }
```

This workaround will do until new version of `embassy-stm` gets released.

Thanks.