embassy-rs/embassy #4537: High nibble of MIDI status byte is duplicated when reading MIDI messages

universalhandle, 2025-08-12:
I adapted a [couple](https://github.com/embassy-rs/embassy/blob/c361c82c41af1697d5c849b817554074e6dd6c8b/examples/stm32f7/src/bin/usb_serial.rs) of Embassy [examples](https://github.com/embassy-rs/embassy/blob/996974e313fa5ec2c7c2d9dd0998fab244c0a180/examples/rp/src/bin/usb_midi.rs) to receive MIDI over USB for an F-7 series STM32 microcontroller. Sending a Middle C from my laptop to the dev board outputs the following:

```
118.646118 [INFO ] data: [9, 90, 3c, 40] (miditest miditest/src/main.rs:158)
118.731353 [INFO ] data: [8, 80, 3c, 40] (miditest miditest/src/main.rs:158)
```

I repeated this several times, generating the MIDI from Ardour (a DAW), some virtual keyboard I installed from the Ubuntu repository, and even by typing hex values into midisnooper. I've no reason to believe there's a problem with ALSA, as MIDI from these sources properly drives commercial synths I have on hand.

The last three bytes of each message are as expected:

| Status byte (hex, dec)                    | Note (hex, dec) | Velocity (hex, dec) |
| ---------------------------------------------- | ---------------------- | -------------------------- |
| Channel 1 Note On (0x90, 144)   | C4 (0x3c, 60)     | (0x40, 64)                 |
| Channel 1 Note Off (0x80, 128)   | C4 (0x3c, 60)     | (0x40, 64)                 |

The first item in `data`, however, was unexpected. After some more investigation, I realized that the first value in `data` is always the four most significant bits of the status byte. (A MIDI status byte is the first in a series which comprise a message. The number of bytes in the message depends on the status byte, which can be thought of as the "type" of the message. Only status bytes may have a 1 in the MSB. The data bytes that follow, each having a 0 in the MSB, complete the message, e.g., which key was pressed, and how hard?)

I don't understand all of Embassy's USB code, but as far as I can tell `MidiClass` doesn't heavily participate in data handling. Primarily its job seems to be to declare and describe the device's interfaces. Its `read_packet` method delegates to the USB driver, which in my case is `embassy-usb-synopsys-otg`.

I have a few questions:

1. Is this behavior intentional? It seems unlikely, but maybe there's something I don't know about writing generalized On-The-Go USB drivers. If it were, I suppose the fix would be to add additional logic to  `MidiClass` to further process data from the driver, rather than to change the driver per se. But, again, this is hard to fathom.
2. I'm not sure if the bug (if it is not in my own code) is in `embassy-usb-synopsys-otg` or deeper. I had trouble following the code past that.
    1. Would it help to post the code I adapted? It's essentially the same as the examples. I suppose it would require someone to have a substantially similar chip to mine to be very useful.
    2. Assuming the bug is not in my code, how is the `out_size` in the endpoint state getting set to 4? I guess I'm also having trouble seeing how the `out_buffer` gets populated.
3. What is the desired scope/trajectory for `MidiClass`? Because the `Note On` and `Note Off` messages were dumped separately, I assumed `MidiClass` was more involved in data processing, i.e., that it had identified the `Note On` bytes as components of a single message and thus grouped them together, then grouping the next set of bytes when it detected the `Note Off` status byte. I suppose my brain substituted `read_message` where the code says `read_packet`. I think that grouping must be a product of timing, though, since when I have my DAW play two notes simultaneously, all the bytes are presented at one time: `[9, 90, 3c, 40, 9, 90, 3a, 40]`. (Note that nibble is duplicated for both `Note On` events, making it very challenging to guess what to do with that second `0x9`.) Assuming we get the duplicated data out of the picture, is grouping bytes into messages (I hesitate to use the word "parsing" for this) something that should be handled in this class, or downstream?

Thanks!

universalhandle, 2025-09-07:
Well, I feel a little silly now. Thank you, @cryptographix, for pointing me in the right direction. The amount of documents associated with the MIDI standard(s) is a bit overwhelming.

At first glance, the design choice is surprising. Obviously I'm no expert, but the leading bit that is unique to the USB implementation appears to be (at least in the handful of events I've looked at) redundant with the rest of the packet. For example, the last four bits in a Note On status byte provide the same information as the first four bits in the packet header (unless channel and cable number are not the same thing), and the first four bits of the status byte (`0x9`) match the Code Index Number, which indeed signifies Note On according to the doc you linked. Hmm, looking forward to digging deeper and seeing if there are events for which this is not the case.

It really helps to be looking at the _correct_ documentation. Thanks again.

closed 2025-09-07 by universalhandle

cryptographix, 2025-09-07:
@universalhandle don't worry .. USB-MIDI is confusing and a bit weird IMHO, and if you look at MIDI 2.0 over USB it gets a lot worse.

Anyway, just to clear up a point you mentioned, jack ID/cable number and midi channel are not the same thing. The idea of cable numbers is to allow you to have multiple virtual midi cables over USB (thus the jack-in / jack-out information received during endpoint enumeration). Each virtual cable (0-15) can have up to 16 channels as per normal midi.

Tn MIDI 2.0, this idea of cable numbers, has been expanded into "UMP Groups", where you can have up to 16 groups, each with 16 channels. But then they add in Function Blocks, Group Terminal Blocks and a whole load of other complexity such as CI discovery (query and enumeration), profiles, etc. The spec makes great reading for a sleepless night.

As for Code Number Index, it is also a different beast to the Status byte, despite having a few values in common. I think the idea was mainly to simplify parsing of System status messages, but as you'll see in MIDI 2.0, that's all been changed so probably didn't work too well :)

Have fun!