embassy-rs/embassy #4558: Order of ADC and Channel creation impacts results

cconstantine, 2025-08-17:
If you move [this line](https://github.com/embassy-rs/embassy/blob/bfe4395b3b9792bb79363e2f32a9ab7bf69bb78d/examples/rp/src/bin/adc.rs#L25) to [here](https://github.com/embassy-rs/embassy/blob/bfe4395b3b9792bb79363e2f32a9ab7bf69bb78d/examples/rp/src/bin/adc.rs#L31), the readings from the ADC will be wrong.

It appears that you need to initialize the ADC before creating the channels, but that isn't obvious from the code and the docs don't say anything about it.  Would it make sense to add something to the docs?

9names, 2025-08-17:
So you're saying that if you do
```rust
let mut p26 = Channel::new_pin(p.PIN_26, Pull::None);
let mut adc = Adc::new(p.ADC, Irqs, Config::default());
```
instead of

```rust
let mut adc = Adc::new(p.ADC, Irqs, Config::default());
let mut p26 = Channel::new_pin(p.PIN_26, Pull::None);
```
it doesn't work properly?

I think just documenting this isn't going to save people from the footgun here.
Probably best if we require a reference to an `Adc` instance to set up Channels so that you can't do it wrong.

9names, 2025-08-17:
This seems to only affect the temperature sensor, regular ADC measurements still occur regardless of order