embassy-rs/embassy #4659: Battery backed RAM

usbalbin, 2025-09-13:
What do you think about adding some nice API for accessing the battery backed RAM found in, for example, the stm32h5 series?

---

I suppose we could always offer some simple API with read/write methods.

However, do you think it would be possible to figure out how to achieve something like this in a safe way?

```rust
enum ReasonForDataReset {
    /// This is the first time we run or the first time after battery power was lost
    BackupRamDisabled,

    /// The struct layout does not seem to match
    ///
    /// `all_bytes` may be used to try to recover if the firmware was updated on purpose
    StructureMissmatch{ all_bytes: &mut[u8] }
}

#[repr(C)]
struct MyStruct {
    foo: u32,
    bar: MyEnum, // <-- Not all bit patterns are valid for this type
}

// This will be run the first time to set the defaults and on struct layout missmatch
let fn_to_setup_defaults = |_reason: ReasonForDataReset| MyStruct { foo: 42 };

// Here we get a mutable reference to the value stored in battery backed ram
let x: &'static mut MyStruct = BackupRam::init::<MyStruct>(p.BACKUP_RAM, fn_to_setup_defaults);

loop {
    x.foo += 1; // These writes will persist resets and loss of vcc
    defmt::info!("I have ran {} iterations since I was programmed/battery was inserted", x.foo);
}
```

How do we ensure
* The user uses the same type with the same layout every time
    * Would it be possible to create a checksum for the types layout and run fn_to_setup_defaults on missmatch or something?
* The type does not contain pointers/references out of the battery backed memory
    * Create a trait for types that are safe to store?
* Large structures does not overflow the stack in fn_to_setup_defaults.

nikvoid, 2025-09-18:
I use custom RAM section and unsafe code for this purpose:
```rust
/// Wrapper for storing persistent value in backup SRAM
///
/// Must be used carefully:
/// - `T` must not have any pointers
/// - `T` must not have `Drop` impl
/// 
/// Should be in `static` in section `.bkpsram`
#[repr(C)]
pub struct BackupSram<T> {
    init: UnsafeCell<MaybeUninit<u32>>,
    data: UnsafeCell<MaybeUninit<T>>,
}

impl<T> BackupSram<T> {
    /// Init magic number
    const INIT: u32 = 0xA7EE7151;
    
    /// No-op initializer
    pub const fn new() -> Self {
        Self {
            init: UnsafeCell::new(MaybeUninit::uninit()),
            data: UnsafeCell::new(MaybeUninit::uninit()),
        }
    }

    /// Overwrite previous value, if it exists
    pub unsafe fn set(&self, f: impl FnOnce() -> T) {
        self.data.get().write_volatile(MaybeUninit::new(f()));
        self.init.get().write_volatile(MaybeUninit::new(Self::INIT));
    }
    
    /// Get reference to current value
    pub unsafe fn get(&self) -> Option<&mut T> {
        if self.init.get().read().assume_init() != Self::INIT {
            return None;           
        }

        Some(self.data.get().as_mut_unchecked().assume_init_mut())
    }

    /// Reset init state and zeroize data
    pub unsafe fn reset(&self) {
        self.init.get().write_volatile(MaybeUninit::zeroed());
        self.data.get().write_volatile(MaybeUninit::zeroed());
    }

    /// Get value and reset init state
    pub unsafe fn take(&self) -> Option<T> {
        if self.init.get().read().assume_init() != Self::INIT {
            return None;           
        }
        
        let ret = Some(self.data.get().read_volatile().assume_init());
        
        self.reset();

        ret
    }
}
```

```ld
MEMORY {
  FLASH (rx)   : ORIGIN = 0x08000000, LENGTH = 896K
  RAM (xrw)    : ORIGIN = 0x20000000, LENGTH = 512K
  BKPSRAM (rw) : ORIGIN = 0x40024000, LENGTH = 4K
}

SECTIONS {
  .bkpsram (NOLOAD): {
    KEEP(*(.bkpsram))
    KEEP(*(.bkpsram.*))
  } > BKPSRAM
}
```

With this approach you just can't use other type, but it is not actually checked for being Plain Old Data. All reads are volatile so compiler (I hope) don't assume it may be optimized in some way.

usbalbin, 2025-11-05:
Thanks for the try :)

Where do I find the source for the tests?

Is it just `embassy/tests/stm32/src/bin/cordic.rs`?

usbalbin, 2025-11-05:
Thanks!

From the log, the BDCR has a value of 0c008213 which seems a bit wonky. That seems to indicate that `BREN` (bit 0) is enabled, yet `enable_backup_sram` is never set in the tests from what I can tell and should therefore default to `false` with BREN being 0.

[test run](https://ci.embassy.dev/jobs/f25a96a6f217)

```
 ERROR teleprobe::client > === nucleo-stm32h563zi out/tests/stm32h563zi/cordic: FAILED: HTTP request failed with status code: 400: Bad Request
 ERROR teleprobe::client > INFO - run_from_ram: true
INFO - flashing program...
INFO - flashing done!
DEBUG - 0.000000 flash: latency=5 wrhighfreq=2
TRACE - 0.000000 BDCR configured: 0c008213
DEBUG - 0.000000 rcc: Clocks { audioclk: MaybeHertz(0), csi: MaybeHertz(0), hclk1: MaybeHertz(250000000), hclk2: MaybeHertz(250000000), hclk4: MaybeHertz(250000000), hse: MaybeHertz(8000000), hse_div_rtcpre: MaybeHertz(0), hsi: MaybeHertz(0), hsi48: MaybeHertz(48000000), lse: MaybeHertz(0), lsi: MaybeHertz(0), pclk1: MaybeHertz(250000000), pclk1_tim: MaybeHertz(250000000), pclk2: MaybeHertz(250000000), pclk2_tim: MaybeHertz(250000000), pclk3: MaybeHertz(250000000), pll1_q: MaybeHertz(250000000), pll2_p: MaybeHertz(0), pll2_q: MaybeHertz(0), pll2_r: MaybeHertz(0), pll3_p: MaybeHertz(0), pll3_q: MaybeHertz(0), pll3_r: MaybeHertz(0), rtc: MaybeHertz(32000), sys: MaybeHertz(250000000) }
WARN - Deadline exceeded!
INFO -   R0: 00000000   R1: 00000000   R2: 20008a8c   R3: 00000000
INFO -   R4: 2009ffe8   R5: 00000000   R6: 00000000   R7: 2009ffe0
INFO -   R8: ffffffff   R9: ffffffff  R10: ffffffff  R11: ffffffff
INFO -  R12: 0000000e   SP: 2009ffd8   LR: 20000357   PC: 20002d26
INFO - XPSR: 69000000
INFO - 
INFO - Stack:
WARN - send_jtag_command 242 failed: SwdApFault
ERROR - Run failed: An ARM specific error occurred.

Caused by:
    0: The debug probe encountered an error.
    1: An error which is specific to the debug probe in use occurred.
    2: Command failed with status SwdApFault.
```

I only have a H523 and H533s and the cordic seems to be unavailable there, so I can not run the cordic test locally to debug... However I'll try to do some more digging on why the BREN bit seems to get set when it should not

xoviat, 2025-11-05:
I just merged #4388, and did not see any issues. So something about these changes is creating a problem.

usbalbin, 2025-11-05:
> do you have access to any of the failing boards?

No, sorry. I have H523 and H533 but they do not have the cordic peripheral.

> I just merged https://github.com/embassy-rs/embassy/pull/4388, and did not see any issues. So something about these changes is creating a problem.

That is strange

usbalbin, 2025-11-05:
I could probably get a H563 but that would probably take a week or so