embassy-rs/embassy #4689: stm32u5: uart read_until_idle() has a mistake

fan2nd, 2025-09-21:
# Description
When I tried the serial loopback test, I sent '1', '2', '3', '4' in sequence. The data returned from the serial port had no issues, but there were duplicate entries in the defmt log. As shown in the figure below.

[image]

# Envirenment
- board: stm32u545re-nucleo
- software: embassy + defmt_rtt + probers
```
#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::{
    bind_interrupts, peripherals,
    usart::{self, Config, Uart},
};
use {defmt_rtt as _, panic_probe as _};

bind_interrupts!(struct Irqs{
    USART1 => usart::InterruptHandler<peripherals::USART1>;
});

#[embassy_executor::main]
async fn main(_spawner: Spawner) -> ! {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let config = Config::default();
    let mut uart = Uart::new(
        p.USART1,
        p.PA10,
        p.PA9,
        Irqs,
        p.GPDMA1_CH0,
        p.GPDMA1_CH1,
        config,
    )
    .unwrap();
    let mut buffer = [0u8; 32];
    loop {
        if let Ok(len) = uart.read_until_idle(&mut buffer).await {
            info!("{}", &buffer[0..len]);
            uart.write(&buffer[0..len]).await.unwrap()
        }
    }
}

```

vDorst, 2025-09-22:
I also noticed see #3900.
I hope I have soon time to look at it myself because I also need it for my project.

fan2nd, 2025-12-06:
It has been solved by pr #4981